asund/esphome-daikin-s21 #21: Support Intelligent Eye

benwa, 2025-06-18:
[Page 15 of the FTXS/CTXS manual](https://backend.daikincomfort.com/docs/default-source/product-documents/residential/manuals/operationmanuals/ftxs09_12lvju-ctxs07lvju-operation-manual.pdf) shows Intelligent Eye.
The [ESP32-Faikin reverse engineer documentation](https://github.com/revk/ESP32-Faikin/blob/main/Manuals/S21.md) does not yet have that information, so more work is required.

asund, 2025-06-22:
If we can't discover support I think we could try to tap into the PIR daughterboard's signal via the S36 / S26 / S25 connector and feed it into a spare GPIO on the ESPHome board. These sensors are very simple and with a bit of esphome sensor filtering I think we could have a much better implementation than Daikin's rigid algorithm:
[image]

Smandurlo, 2026-01-30:
@asund so I think I have nothing from my units about intelligent eye

closed 2026-10-29 by asund