asund/esphome-daikin-s21 #42: No connection after flashing

txitxo0, 2025-07-18:
Hi there, really good job taking the oldated project to keep it up to date and allowing us using it, thank you :)

I am moving now from the [joshbenner](https://github.com/joshbenner) project to yours, but I am having a connection problem once the new poject is flashed. 

I started only changin the repo and I will get the new properties available to play with after having a working climate.

So that, my yaml definition is like below, once the code is flashed with no error, the api ir not availble and the IP (in my case 192.168.3.88) not reachable, but no useful logs found:

`WARNING Can't connect to ESPHome API for climate-tvroom-esp-daikin @ 192.168.3.88: Error while starting connection: Starting connection cancelled (APIConnectionCancelledError)`

could you help me?

```yaml
substitutions:
  devicename: climate-tvroom-esp-daikin
  upper_devicename: Aire Salon

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

esphome:
  name: $devicename

api:
ota:
  platform: esphome
web_server:
  port: 80

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: $devicename
    password: !secret ap_password
  manual_ip:
    static_ip: 192.168.3.88
    gateway: 192.168.3.1
    subnet: 255.255.255.0

logger:
  baud_rate: 0  # Disable UART logger if using UART0 (pins 1,3)

external_components:
  - source: github://asund/esphome-daikin-s21@main
    components: [ daikin_s21 ]

uart:
  - id: s21_uart
    tx_pin: GPIO21
    rx_pin: GPIO20
    baud_rate: 2400
    data_bits: 8
    parity: EVEN
    stop_bits: 2

# The base UART communication hub.
daikin_s21:
  tx_uart: s21_uart
  rx_uart: s21_uart

climate:
  - name: $upper_devicename
    platform: daikin_s21
    visual:
      temperature_step: 0.5
    #supports_humidity: true # If your unit supports humidity, it can be reported in the climate component
    # Optional HA sensor used to alter setpoint.
    room_temperature_sensor: room_temp  # See homeassistant sensor below
    # setpoint_interval: 300s

# Optional additional sensors.
sensor:
  - platform: daikin_s21
    inside_temperature:
      name: $upper_devicename Inside Temperature
    outside_temperature:
      name: $upper_devicename Outside Temperature
    coil_temperature:
      name: $upper_devicename Coil Temperature
    fan_speed:
      name: $upper_devicename Fan Speed

  - platform: homeassistant
    id: room_temp
    entity_id: sensor.netatmo_smart_thermostat_current_temperature #sensor.zb_salon_sensor_temperature
    unit_of_measurement: °C
```

asund, 2025-07-19:
First off, thanks for following and helping making it better and sorry for your interruption in service. I recognize from my own experience it's frustrating when you have to open a unit up to take out the microcontroller and reflash via USB...

I can think of two possible areas to investigate:

1. I've screwed up some handling with the external temperature sensor handling. This is not a feature I use but can hopefully take a look this weekend at configuring a sensor to see if it's causing it to go off the rails. Disabling this in your config to see if it begins functioning would be something you could try in the meantime.
2. Some of the model detection code I've written or attempts at compatibility for hardware I don't have is at fault. This will be harder for me to debug.

Do you have a laptop a setup where you could get logs over the USB connection from the device as it boots? It may help narrow things down. Please enable additional debug logging if this works for you:
```
daikin_s21:
  tx_uart: s21_uart
  rx_uart: s21_uart
  debug_comms: true
  debug_protocol: true
```

asund, 2025-07-21:
Thanks for checking that, despite the hassle. It's not the external temperature sensor and your system is pretty similar to mine so those two theories are less promising.

One major area of change between the two projects was the serial command state machine I added to run communication in the background (and ironically enough, keep my system responsive enough to stay on the wifi). I've been running the IDF framework instead of Arduino ever since finding it worked better for one project a while back. Their APIs are subtlety different though, as the original author had to resort to having two UARTs in order to invert one of the pins. I think this is fully supported with the native ESPHome pin configuration code when using the IDF framework.

From experience when the serial code goes off the rails, the system can easily lock up or crash. It'd be very easy for a difference in API between the two frameworks to result in a hang trying to read a byte that's not coming forever.

I can test this framework theory pretty easily and it sounds like I have better access for debugging. I'll try to get to it early this week.

asund, 2025-07-25:
Sorry, it's been busy for me lately but I do still intend to get to this. If you're impatient I can suggest trying the ESP-IDF framework, it's what I use and can confirm works for me.

closed 2025-07-28 by txitxo0