asund/esphome-daikin-s21 #80: Problem with Faikin board

tmarschner, 2025-09-30:
Hi, Newbie here. Not sure if this is a bug or if I just don't know what I'm doing. I have a new/recent Faikin board with an ESP32-S3-mini that works fine with the Faikin firmware to control my Daikin ceiling unit. This board appears to invert both tx and rx (GPIO48 and GPIO34, respectively.) When I flash with the ESPHome firmware I see the errors shown below. I'm not sure how to proceed. I tried all combinations of inverted true/false in the uart config but that didn't help. 

```[07:52:43.784][D][daikin_s21.serial:148]: Tx: F8
[07:52:44.037][W][daikin_s21:848]: Timeout waiting for response to F8
[07:52:44.290][D][daikin_s21.serial:148]: Tx: FY00
[07:52:44.545][W][daikin_s21:848]: Timeout waiting for response to FY00
[07:52:44.799][D][daikin_s21.serial:148]: Tx: F8
[07:52:45.050][W][daikin_s21:848]: Timeout waiting for response to F8
[07:52:45.305][D][daikin_s21.serial:148]: Tx: FY00
[07:52:45.557][W][daikin_s21:848]: Timeout waiting for response to FY00
```

The relevant portion of my yaml config file looks like this:

```
esphome:
  name: brdaikin
  friendly_name: "Bedroom Daikin"
  min_version: "2025.8"
  devices:
    - id: daikin_outdoor
      name: "Daikin Compressor"

esp32:
  variant: esp32s3 
  framework:
   type: esp-idf

logger:
  baud_rate: 0  # Disable UART logger if using UART0 (pins 1,3)

external_components:
  - source: github://asund/esphome-daikin-s21@main
    components: [ daikin_s21, split_uart ]
    refresh: 0s

uart:
  - id: s21_uart
    tx_pin: 
      number: GPIO48
      inverted: true
    rx_pin: 
      number: GPIO34
      inverted: true
    baud_rate: 2400
    data_bits: 8
    parity: EVEN
    stop_bits: 2

daikin_s21:
  uart: s21_uart
  debug_protocol: true
  debug_comms: true

```
Thanks in advance for any help.

tmarschner, 2025-10-15:
I do have a logic analyzer but I've only used it a few times for messing with SPI signals. I am a real newbie and am just getting started with esp32 (and electronics in general.) (Post-retirement hobby) I did try looking at the tx and rx leads on the connector with the logic analyzer but didn't see anything and wasn't sure if it wasn't working or i didn't set up the logic analyzer correctly. (board was running your software and being powered using a TC2030 connector and cable.) I will try reinstalling the Faikin software which I know works with my Daikin unit and work out the logic analyzer setup and settings (at least for the transmit from the Faikin board). If I can get this to work I will switch to your software and look for transmit data. Does this make sense? Any other tips would be appreciated.

asund, 2025-10-15:
Nevermind about the logic analyzer! @tsunglung's #87 should resolve this for you and allow pin inversion to be configured properly. I've merged to main. Give it a shot and if it works for you I'll clean up some uart related settings, including the split_uart component.

tsunglung, 2025-10-15:
My yaml configuration for your reference
The faikin works
```
uart:
  id: s21_uart
  tx_pin:
    number: GPIO48
    inverted: true
  rx_pin:
    number: GPIO34
    inverted: true
  baud_rate: 2400
  data_bits: 8
  parity: EVEN
  stop_bits: 2

# The base UART communication hub.
daikin_s21:
  uart: s21_uart
```

logs
```
[00:37:31.099][D][daikin_s21:949]: Protocol: 0.0  ModelV0: 35E4  ModelV2: FFFF
[00:37:31.108][D][daikin_s21:965]: ModelInfo: N  VSwing: Y  HSwing: Y  Humid: N  Fan: Y
[00:37:31.108][D][daikin_s21:981]: Mode: OFF  Action: IDLE  Preset: NONE  Demand: 0
[00:37:31.108][D][daikin_s21:987]: Fan: 1 (0 RPM)  Swing: HORIZONTAL
[00:37:31.108][D][daikin_s21:992]: Setpoint: 27.0C  Target: 27.0C  Inside: 29.5C  Coil: 26.0C
[00:37:31.108][D][daikin_s21:1000]: Cycle Time: 2233ms  UnitState: 0  SysState: 40
```

asund, 2025-10-16:
Very strange, it appears there's no pull up function on GPIO34 on the ESP32-S3. Somehow it works for the Faikin code though. Is main.cpp being rebuilt by ESPHome when you change the invert settings?

@tsunglung, I am working on a text_sensor component that should make collecting debug queries easier. I didn't plan on making another wiki here as the Faikin project has been the home of most of the reverse engineering of the protocol. If I need to document more about the ESPHome integration I could see it happening, but for now I think it's best to combine forces with them as they have more users and hardware and have welcomed the additional documentation I've contributed.

tmarschner, 2025-10-16:
Yes, main.cpp is being recompiled when I change the invert settings. The version of the board I have (https://github.com/revk/ESP32-Faikin/commit/54dcb000f7bda7bc43a0248b6b7e3f904d46b386) doesn't have that testpoint.  So i tried very carefully with a needle probe to connect the logic analyzer to pin 6 (rx on esp32) of the mosfet IC. When I try this with the board running the Faikin software I do see valid S21 protocol data on the esp32 rx pin, although it's inverted from the pin on the daikin side of the mosfet (pin 2). When I do this exact same thing with the board running esphome-daikin-s21 software I don't see any data on pin 6. Based on [this document](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/gpio.html#_CPPv418gpio_set_pull_mode10gpio_num_t16gpio_pull_mode_t) it looks like GPIO34 (rx) doesn't support pullup. I defer to people who understand this stuff to draw a conclusion. ;)

asund, 2025-10-17:
No signal on the Rx pin is very strange.

The assumed esp32-s3-devkitc-1 board by the esp32s3 variant setting... https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-devkitc-1/user_guide_v1.1.html#hardware-reference shows this pin isn't broken out. Maybe this affects ESPHome's internal pin mapping stuff. I'll try to dig into their code over the next little while and figure this out.

Some assumption about PSRAM being octal (that chip is quad) might also affect that pin, but it should always be an input...

tsunglung, 2025-10-17:
I think that no signal on the Rx pin was caused by incorrect S21 Tx signal. 
There is no PSRAM in Faikin board. GPIO34 can be used as Rx pin.
The GPIO32 was connected to transistor, No matter pull up or not, it is not effect on S21.

tmarschner, 2025-10-17:
I was confused because I thought the Daikin already pulled up the pin but I didn't consider the not connected state. And I didn't consider that pull-ups can be "weak". Thanks!

closed 2025-10-25 by asund